electrichasgoneaudi/etron-issues #7: Mirror camera fogs

etrondriver, 2020-01-11:
Todo: Add descripton

Hebjag, 2022-02-20:
Anyone aware of class action law suits against Audi re this? If I know some owners have returned their vehicles but just checking if any law firm has taken this on as yet.

steviestainz, 2022-02-20:
I’d love to know too. I’m on my 3rd set of cameras and each time they’re as bad as the last. The first set never heated at all, second set were meant to be an improved design but were no different. Finally the latest suffer only fogging during narrow pressure/temperature range between 4-10degC but the rear view demister does warm the camera. See difference below after 2 mins. However I would like to keep the vehicle for longer due to low mileage but can’t with any confidence unless Audi resolve the problem once and for all and guarantee future parts availability.
[image]

[image]