electrichasgoneaudi/etron-issues #8: Virtual Cockpit is reset

etrondriver, 2020-01-11:
Owners report that Virtual Cockpit is reset to the default mode. 

TODO: Add more description

pultzner, 2020-01-22:
Yes, this happens to me also without known reason

deltamik, 2020-02-02:
VC reset #1

markusv, 2020-02-03:
Yes this happened to me as well. Møller arrogantly suggested I had to use the wrong key, since this is stored in the user profile. Anyhow this stopped happening by itself. I have no idea why

pultzner, 2020-02-03:
I only use digital key and not physical key and it still happens

markusv, 2020-02-03:
I am 100% sure that I dit not use the wrong key. The seats position is also stored with the key. And I can't even get in the car with the seats in my wifes position. So that is just Møller always blaming the user. they told me they would check with the factory. But I never heard back from them. Since this stopped happening I have not followed them up on this issue.

RemiProvost, 2020-02-25:
I also happends for me now and then. Very frustrating, especially because I can’t change back to my preferences until the car car has been fully turned off and “restarted”.
It simple just doesn’t respond to input changes while in this bad behavior mood 😅

I always use my own key, my wife has the other.

wataabirs, 2022-04-26:
I had this problem also, but I found a possible workaround: after adjusting the VC to the e-tron mode I have saved my seat position with the seat memory function and now the VC stays e-tron mode (for a couple of weeks already). Before this it reverted to the Standard mode every couple of hours/days.