electrichasgoneaudi/etron-issues #40: Regeneration fails randomly

strilen-etron, 2020-02-02:
At random times the regeneration dont work and only conventional brakes work.
Typically shortly after starting to drive after a night in the garage.
By this you have to "step harder" on the brake pedal to get the "same effect".
Have been surprised by this a few times in situations. Asked for workshop appointment to review this.

markusv, 2020-02-02:
I read somewhere this is a feature to prevent  brakes from rusting. After I read this I started noticing that my etron does this almost every time on my way out of the garage. I have not noticed a difference in the  difference in performance though..

TheTechArch, 2020-02-02:
I am pretty sure at is as designed. I have a garage at a top of a hil, and going Down that hill regeneration never Works. Typical starts working after 10-15 Seconds of braking.

AJRS3563, 2020-02-19:
Regen will not happen if:
1. Battery is at 100% charge (no room / time to capture the energy)
2. Battery is not in the right temperature range to capture the energy
3. Car decides it's time to wipe the rust off the rotors