electrichasgoneaudi/etron-issues #56: Is it possible with one pedal driving on e-tron

etrondriver, 2020-07-06:
Question: Is it possible with one pedal driving on e-tron? 

Background: Many less advanced EVs support one pedal driving meaning that when you lift the foot from the gas pedal it will activly reduce the speed using regeneration. In this way you never need to use the brake pedal (you can if it is an emergency).   That is the reason they call it "one-pedal-driving"

Answer:  Audi has not made it possible to enable one pedal driving permanent. They rather want their drivers to free roll instead of autmatic reducing speed when you lift the foot of the gas pedal. One pedal driving is less effecient since you loose energy compared to free rolling since you only will get 80% of the energy back to the battery. You is also forced to have the foot on the pedal at a specific position to free roll. 

If you have set recuperation to manual you can every time you start a drive, set recuperation to max with the steering wheel pedals and then it will regenerate everytime you lift your foot of the break. This is not recomended for experienced drivers.

MEntOMANdo, 2021-09-11:
This write-up is nice, but I wholeheartedly disagree with the notion that it shouldn't be a feature. I disagree also with the statement that for best results one should coast. That assumes that electric efficiency is the utmost and primary goal. If indeed that is the goal, then Audi should also remove sport mode, and normal drive mode, and leave us just with eco mode, because after all that is the most efficient. Further, we can get rid of all the air suspension options, too. Simply have it set to automatic. 

Obviously I'm being facetious. One pedal driving is clearly a user preference, and one that should be supported and respected. All of the modern EVs do in fact provide one pedal drive modes nowadays, and even Tesla retrofitted the feature for those cars which had the appropriate drive motors to handle it.

I hope Audi adds true one pedal driving.

VFXpro, 2023-10-22:
This is the 2nd time I passed on buying an Audi since they don't offer automatic 1 pedal driving.  They current approach of setting it up manually each time is too annoying.  I owned 4 Tesla since 2011 and every car that doesn't offer 1 pedal driving looses me as a customer. (Lexus, Audi, etc...).   But Polestar, Tesla, etc... are now my only choices.