electrichasgoneaudi/etron-issues #57: What is the differences between the different model years of e-tron

etrondriver, 2020-07-12:
Question: What is the differences between the different model years of e-tron

Answer:

MY2019 - > 2020
- Lidar radar is gone on 2020 model
- It is possible to adjust charging target on 2020 models in app.
- vin nr has a L as 10 digit for 2020 model example:  WAUZZZGE0**L**B028888

steviestainz, 2020-07-12:
I wonder why the LIDAR has been removed? I always felt that the majority of hardware required for semi autonomous driving was in the e-tron already and it was a shame that Audi hadn’t made more of that potential in comparison to Tesla’s effort. Has it been replaced with something better?

etrondriver, 2020-07-13:
@steviestainz i understand that the reason for that was it was planned to be used with Autonomous driving at level 3, but that Audi did not belive it would be able to add in a reasonable time because of regulation. But I see from the e-tron S 2021 models that the Lidar is back again. I am not sure it will be on the production models.