electrichasgoneaudi/etron-issues #111: Broken/defech front motor

deltamik, 2021-07-19:
Front motor on my e-tron 55 2019 was broken. I identified this by a very strange sound from the front axel sounding like a bad electric dynamo/motor. Really noticable both from inside and outside. Scary thing is that there was no error message and I had to identify this myself and also wait 4 weeks for an inspection at Møller bil. It took an additional 10 weeks to fix.

I have videos of the sound while driving.

deltamik, 2021-07-19:
Videos attached

Srekrek, 2021-11-11:
Front motor of my e-tron 55 2019 was broken after about 90000km. I got a several other errors, and after parked on a small slope, it didn't "start". After towing the car for a couple of 100 m, it did start, and I drove a couple of 100 km to the Audi workshop. repair took about 1 week.
Now, after about 107000 km it got it's software update / callback 93M3 (also took some days because of a temporarily pause of the rollout from Germany). They noticed that my back e-motor was producing a lot more noise as usual (I did notice that, but mainly at low driving speeds). They are now going to replace the back motor as well in a couple of weeks. It still drives fine, but they will do this to prevent a motor brake down in the nearby future.